RoboticsBrno/SmartLeds #1: Problem with multiple ledstrip

kakou-fr, 2017-07-15:
Hi,

I have a problem when I want to use 2 (or more) ledstrip, for example with : 
`#include "LearningKit.h"
#include <SmartLeds.h>

const int LED_COUNT = 72;
const int DATA_PIN1 = 12;
const int DATA_PIN2 = 13;
const int CHANNEL1 = 0;
const int CHANNEL2 = 1;

SmartLed leds1( LED_WS2812B, LED_COUNT, DATA_PIN1, CHANNEL1, DoubleBuffer );
SmartLed leds2( LED_WS2812B, LED_COUNT, DATA_PIN2, CHANNEL2, DoubleBuffer );

void setup() {
        Serial.begin(921600);
        delay(200);
        //Initialize the led pin
        pinMode(LED_BUILTIN, OUTPUT);
}

uint8_t hue;
void showGradient() {
        hue++;
        // Use HSV to create nice gradient
        for ( int i = 0; i != LED_COUNT; i++ ) {
                leds1[ i ] = Hsv { static_cast< uint8_t >( hue + 30 * i ), 255, 255 };
                leds2[ i ] = Hsv { static_cast< uint8_t >( hue + 30 * i ), 255, 255 };
        }
        leds1.show();
        leds2.show();
        // Show is asynchronous; if we need to wait for the end of transmission,
        // we can use leds.wait(); however we use double buffered mode, so we
        // can start drawing right after showing.
}

void showRgb() {
        for ( int i = 0; i != LED_COUNT; i++ ) {
                leds1[ i ] = Rgb { 255, 0, 0 };
                leds1.show();
                leds2[ i ] = Rgb { 255, 0, 0 };
                leds2.show();
        }
        delay(500);
        for ( int i = 0; i != LED_COUNT; i++ ) {
                leds1[ i ] = Rgb { 0, 255, 0 };
                leds1.show();
                leds2[ i ] = Rgb { 0, 255, 0 };
                leds2.show();
        }
        delay(500);
        for ( int i = 0; i != LED_COUNT; i++ ) {
                leds1[ i ] = Rgb { 0, 0, 255 };
                leds1.show();
                leds2[ i ] = Rgb { 0, 0, 255 };
                leds2.show();
        }
        delay(500);
}

void loop() {
        Serial.write("Hello2!\n");
        digitalWrite(LED_BUILTIN, HIGH);
        if ( millis() % 10000 < 5000 )
                showGradient();
        else
                showRgb();
        delay( 500 );
        digitalWrite(LED_BUILTIN, LOW);
        delay( 50 );
}
`

the board crash on the show() call

`
PincabLedStrip.ino
LedStrip.h
test.ino
LedStrip.cpp
 1
25
26
27
28
29
30
31
32
33
34
35
36
37
38
39
40
41
42
19
20
21
22
23
24
13
14
15
16
17
18
 7
 8
 9
10
11
12
 1
 2
 3
 4
 5
 6

                leds1[ i ] = Hsv { static_cast< uint8_t >( hue + 30 * i ), 255, 255 };
                leds2[ i ] = Hsv { static_cast< uint8_t >( hue + 30 * i ), 255, 255 };
        }
        leds1.show();
        leds2.show();
        // Show is asynchronous; if we need to wait for the end of transmission,
        // we can use leds.wait(); however we use double buffered mode, so we
        // can start drawing right after showing.
}

void showRgb() {
        for ( int i = 0; i != LED_COUNT; i++ ) {
                leds1[ i ] = Rgb { 255, 0, 0 };
                leds1.show();
                leds2[ i ] = Rgb { 255, 0, 0 };
                leds2.show();
        }
        delay(500);

uint8_t hue;
void showGradient() {
        hue++;
        // Use HSV to create nice gradient
        for ( int i = 0; i != LED_COUNT; i++ ) {
void setup() {
        Serial.begin(921600);
        delay(200);
        //Initialize the led pin
        pinMode(LED_BUILTIN, OUTPUT);
}
const int CHANNEL1 = 0;
const int CHANNEL2 = 1;

SmartLed leds1( LED_WS2812B, LED_COUNT, DATA_PIN1, CHANNEL1, DoubleBuffer );
SmartLed leds2( LED_WS2812B, LED_COUNT, DATA_PIN2, CHANNEL2, DoubleBuffer );

#include "LearningKit.h"
#include <SmartLeds.h>

const int LED_COUNT = 72;
const int DATA_PIN1 = 12;
const int DATA_PIN2 = 13;

Hello2!
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40081ac5  PS      : 0x00050034  A0      : 0x00040021  A1      : 0x3ffc0600
A2      : 0x8302304c  A3      : 0x3ffc0cd0  A4      : 0x40081b05  A5      : 0x3ffc05e0
A6      : 0x3ffc1c8c  A7      : 0x3ff56000  A8      : 0x00000001  A9      : 0x400855a8
A10     : 0x00000000  A11     : 0x00060e20  A12     : 0x00000003  A13     : 0x01000000
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000007  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400d1b4c  LEND    : 0x400d1b6b  LCOUNT  : 0x00000000

Backtrace: 0x40081ac5:0x3ffc0600 0x00040021:0x00000000

CPU halted.`

yaqwsx, 2017-07-26:
It seems the interrupt handler is shared and therefore when it is overwritten by the second string, it crashes as the interrupt is triggered when no data are ready.

I don't have a hardware laying around, but I will try to look at it during the weekend.

kakou-fr, 2017-07-27:
OK thanks for reply

kakou-fr, 2017-07-31:
In the interrupt handler, you use "RMT.int_clr.ch0_tx_end = 1;"
But it's not a problem if the channel is not the channel 0 ?

yaqwsx, 2017-08-06:
Hi, I am sorry it took me so long, however I didn't have any hardware to test it on laying around.
The bug is fixed now. The reason for it was as I stated in my previous post - there is only one interrupt handler and when you initialise the second strip, interrupt handler is overwritten and therefore interrupts for the first strip are never cleared and it causes WDT to overflow.

closed 2017-08-06 by yaqwsx

kakou-fr, 2017-08-07:
Thanks a lot