RoboticsBrno/SmartLeds #10: Do I need to use double buffer when I use show() with APA102?

JarekParal, 2018-05-06:
https://github.com/RoboticsBrno/SmartLeds/blob/b94f1e013345dc67d699d1d069abb4b2c35368aa/examples/basicExample/basicExample.ino#L26

I expected that I need also double buffer. Is it true?

JarekParal, 2018-05-06:
You need to use double buffer, when you have show() without waiting to finish the previous transmission.

closed 2018-05-06 by JarekParal

yaqwsx, 2018-05-06:
Nope.
Show is asynchronous -> the function returns immediately, however the frame buffer is being accessed even show has returned. It means, that if you write in the buffer, you can corrupt the data. There are two ways to resolve it:
- call wait before writing new data to buffer
- or you double buffering - there will be two buffers internally and once you call show, the buffers are swapped -> you can write new data immediately as the background tasks are operating on one buffer and you access another buffer. 

```
// Single buffer
leds.show();
leds[0] = Rgb{ 0, 0, 0 }; // Wrong!
// -----------------------------------
leds.show();
leds.wait();
leds[0] = Rgb{ 0, 0, 0 }; // Correct

// Double buffer
leds.show();
leds[0] = Rgb{ 0, 0, 0 }; // Correct
leds.show(); // Wrong! Cannot be sure that previous show is over
// -------------------------------------
leds.show();
leds[0] = Rgb{ 0, 0, 0 }; // Correct
leds.wait();
leds.show(); // Correct
```