RoboticsBrno/SmartLeds #11: Invalid Pixel in data stream

Kyklas, 2018-06-06:
Hi,

I am using this library with https://github.com/OttoWinter/esphomelib to control WS2812B LED strip.

I am facing a particular case where every 4th and 5th LED in the strip show different colors. The funny thing is that the pattern repeats.

00FF00 00FF00 00FF00 00FF00 **80FF00** **0007F00** 00FF00 00FF00 **80FF00** **0007F00** 00FF00 0FF00

I have used a Saleae logic analyzer and a custom analyser to decode the pulses. The timing are correct, I updated them based on https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/ and what I have sampled from the WS2812B chip.

The invalid pixel does not always show. 

I am attaching the output of the ESP32, the intended color is 00FF00. I configured for 85 pixel but 88 are generated, I am guessing that the additional 3 pixel are to complete a 32 block of pulses.
[output.log](https://github.com/RoboticsBrno/SmartLeds/files/2074882/output.log)

Kyklas, 2018-06-06:
Testing the smartled lib alone, the data stream is correct with only 85 pixel sent.

Noticed 88 pixel are sent when the issue occurs.

Kyklas, 2018-06-06:
Laying out the pixel data in a 64 bit/pulse, the first octet of a 32 bit/pulse block is the one showing an error.

0 0 F F 0 0 0 0 F F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 0 0 
7 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
8 0 F F 0 0 0 0 7 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 0 0 
7 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
8 0 F F 0 0 0 0 7 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 0 0 
7 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
8 0 F F 0 0 0 0 7 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 0 0 
0 3 0 0 0 0 F F 0 0 0 0 F F 0 0 
8 0 F F 0 0 0 0 0 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 0 0 
7 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
F 0 F F 0 0 0 0 1 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 C 0 F F 0 0 0 0 
7 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
8 0 F F 0 0 0 0 1 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 0 0 
3 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
C 0 F F 0 0 0 0 7 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 E 0 F F 0 0 0 0 
3 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
8 0 F F 0 0 0 0 7 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 0 0 
3 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
8 0 F F 0 0 0 0 7 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 0 0 
7 F 0 0 0 0 F F 0 0 0 0 F F 0 0 
8 0 F F 0 0 0 0 7 F 0 0 0 0 F F 
0 0 0 0 F F 0 0 8 0 F F 0 0 F F 
0 0 0 0 F F 0 0 0 0 0 0 F F 0 0

Kyklas, 2018-06-07:
The interrupt sometimes comes too late. The data is being sent when the interrupt writes the data.
Late Interrupt :
[image]
Interrupt on time :
[image]

To avoid tight timing on the interrupt processing, the block size is double, using the next channel block.

yaqwsx, 2018-06-07:
Thank you for you deep analysis - when you submitted the issue I was suspecting late interrupts. However I wasn't able to reproduce them on my HW so far. Do you have a minimal example I could try on my own?

yaqwsx, 2018-11-28:
It seems that the FastLED library implemented RMT driver in the recent release. As the library is better maintained and offers more features I advise you to use it as I wasn't able to reproduce your issue and debug it.