RoboticsBrno/SmartLeds #15: shouldn't wait() give back the semaphore ?

Maigre, 2019-04-12:
Hi,
The public wait() method (SmartLeds.h // line 157) try to Take the semaphore in order
to ensure that transmission ended. 

But once wait() took the semaphore, shouldn't it give it back immediately ? 

As i understand it right now if you call the wait() method,
then once the transmission is over, the semaphore is taken but never given back which freeze the whole system, right ?

If i am correct, in SmartLeds.h // line 158

**xSemaphoreTake( _finishedFlag, portMAX_DELAY );**     

should be followed with

**xSemaphoreGive( _finishedFlag );**

Thanks !

Best,

yaqwsx, 2019-04-12:
Yes, your are right. Did you already make a fix? If so, could you send me a PR?

yaqwsx, 2019-04-14:
Fixed in e652cd0

closed 2019-04-14 by yaqwsx