RoboticsBrno/SmartLeds #16: Why APA?

AlexFundorin, 2019-08-22:
I 'm not using apa method, just an single led connection with 3 "+ data -" wires.
Yet, upon compilation, I get the following errors:

```
C:/users/fundorin/.platformio/lib/SmartLeds_ID1740/src/SmartLeds.h: In constructor 'Apa102::ApaRgb::ApaRgb(uint8_t, uint8_t, uint32_t, uint32_t)':
C:/users/fundorin/.platformio/lib/SmartLeds_ID1740/src/SmartLeds.h:296:26: error: 'Apa102::ApaRgb::r' will be initialized after [-Werror=reorder]
         uint8_t v, b, g, r;
                          ^
C:/users/fundorin/.platformio/lib/SmartLeds_ID1740/src/SmartLeds.h:296:23: error:   'uint8_t Apa102::ApaRgb::g' [-Werror=reorder]
         uint8_t v, b, g, r;
                       ^
C:/users/fundorin/.platformio/lib/SmartLeds_ID1740/src/SmartLeds.h:280:9: error:   when initialized here [-Werror=reorder]
         ApaRgb( uint8_t r = 0, uint8_t g = 0, uint32_t b = 0, uint32_t v = 0xFF)
         ^
C:/users/fundorin/.platformio/lib/SmartLeds_ID1740/src/SmartLeds.h:296:23: error: 'Apa102::ApaRgb::g' will be initialized after [-Werror=reorder]
         uint8_t v, b, g, r;
                       ^
C:/users/fundorin/.platformio/lib/SmartLeds_ID1740/src/SmartLeds.h:296:20: error:   'uint8_t Apa102::ApaRgb::b' [-Werror=reorder]
         uint8_t v, b, g, r;
                    ^
C:/users/fundorin/.platformio/lib/SmartLeds_ID1740/src/SmartLeds.h:280:9: error:   when initialized here [-Werror=reorder]
         ApaRgb( uint8_t r = 0, uint8_t g = 0, uint32_t b = 0, uint32_t v = 0xFF)
```

And my code contains only this info, regarding smartleds lib:

```
#include <SmartLeds.h>
// LED
const int LED_COUNT = 1;
const int DATA_PIN = 4;
const int CHANNEL = 0;

SmartLed leds(LED_WS2812, LED_COUNT, DATA_PIN, CHANNEL, DoubleBuffer);
```

yaqwsx, 2019-08-22:
It is a bug that was recently fixed (3f66629b49f75d6eda0feb2b8ac513e396440b65), however, my colleague forgot to increase the version number and therefore, the fix was not propagated to the repository. I have increased the version number, but it can take up to 24 hours to propagate in the repository.

Meanwhile, you can add `git@github.com:RoboticsBrno/SmartLeds.git` instead of `SmartLeds` in your platformio.ini as a dependency.

yaqwsx, 2019-08-22:
The fix already present in the repository. Please confirm it works.

AlexFundorin, 2019-08-22:
I can partially confirm, since I've DL the repository, instead of adding some extra lines to the ini file.
So, the errors are disappeared. Yet, I couldn't test the lib itself since it was a surprise for me that my led board is only working with 5V logic, while I'm using 3V3 board in my current project.

closed 2019-08-22 by AlexFundorin

yaqwsx, 2019-08-22:
WS2812 indeed require at least 5V power supply and unreliably work with 3.3V logic signals.