RoboticsBrno/SmartLeds #17: array of leds with a point method

julienhogert, 2019-10-17:
Hello ! Thanks for the job you did.
I tried severals library, and yours is the only one I found, working with 19,18,26,23,5 pins, at the same time. I tried to use the implement of RMT of Fastled, but it didn't work so I wanted to use yours. I'm just facing one problem, I don't find a way to work with the library with a pointer array of smartleds. (It's not a necessity, but just more convenient). There is something too "high-level" for me.  how is possible than a Instance of a class can be igual to a another instance of another class? (Smartled class = rgb class?)
Thanks for your help and the good job you did !

This is the code I used :
`#include <Arduino.h>
#include <SmartLeds.h>

#define NB_STRIPS 4
#define NB_LEDS_PER_STRIP 6
byte leds_pins[]={19,18,26,23,5};

// SmartLed -> RMT driver (WS2812/WS2812B/SK6812/WS2813)
SmartLed *leds[3];

void setup() {
  Serial.begin(9600);  
  for(int i=0;i<NB_STRIPS;i++){
    leds[i]= new SmartLed ( LED_WS2812, NB_LEDS_PER_STRIP, leds_pins[i], i, DoubleBuffer );
}
}

uint8_t hue;
void showGradient() {
    hue++;
    // Use HSV to create nice gradient
   for(int i=0;i<NB_STRIPS;i++){
   for ( int j = 0; j != NB_LEDS_PER_STRIP; j++ ){
        //leds[i][ j ] = Hsv{ static_cast< uint8_t >( hue + 30 * j ), 255, 255 }; NOT WORKING !
        //(*leds)[i][ j ] = Hsv{ static_cast< uint8_t >( hue + 30 * j ), 255, 255 }; // NOT WORKING ! CRASH THE ESP32
        //(*leds)[i][ j ] = Rgb{ 255, 0, 0 }; NOT WORKING !
        }
    leds[i]->show();
    }
}

void loop() {
    Serial.println("New loop");
    showGradient();
    delay( 20 );
}`

yaqwsx, 2019-10-17:
NB_STRIPS is equal to 4, however, your array leds has only 3 elements. Isn't that the whole problem? Please, try to change size of the array to NB_STRIPS and check if the problem still appears.

Also, the correct way to access pixels in your case is `(*leds[i])[j]`.

julienhogert, 2019-10-17:
Yes ! I'm sorry, it's a stupid mistake ! But it didn't change anything. 
I put some "print messages" to see where it was not working / I tried to add some delay, just in case, but it didn't work neither. 

`void showGradient() {
    hue++;
    // Use HSV to create nice gradient
   for(int i=0;i<NB_STRIPS;i++){
   for ( int j = 0; j <NB_LEDS_PER_STRIP; j++ ){
        //(*leds)[i][ j ] = Hsv{ static_cast< uint8_t >( hue + 30 * j ), 255, 255 }; // NOT WORKING ! CRASH THE ESP32
        (*leds)[i][ j ] = Rgb{ 255, 0, 0 }; // NOT WORKING ! CRASH THE ESP32
        Serial.printf("led[%i][%i] / pin %i \n",i,j,leds_pins[i]);
        }
    leds[i]->show();
    //delay(20);
    }
}` 

In the serial monitor, we can see that the program crash at the moment it change the led strip.

`New loop
led[0][0] / pin 19 
led[0][1] / pin 19 
led[0][2] / pin 19 
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0dd1  PS      : 0x00060330  A0      : 0x800d0e6f  A1      : 0x3ffb1f60  
A2      : 0x000000ff  A3      : 0x00000001  A4      : 0x00000034  A5      : 0x3ffbfcfc  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffbfcf8  A9      : 0x3ffbebe1  
A10     : 0x000000ff  A11     : 0x00ffffff  A12     : 0x3ffb8500  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ff56800  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x000000ff  LBEG    : 0x400d0c4c  LEND    : 0x400d0c6e  LCOUNT  : 0x00000000  
`

If you have another solution to avoid this, or another way to program it, I'll appreciate it ! Thanks !! Julien

julienhogert, 2019-10-17:
I also tried to change the pins to see if something was wrong, there, but it didn't change anything !

yaqwsx, 2019-10-18:
I don't have ESP32 laying around, I will try to look at it and replicate your problem (hopefully) by the end of the week.