RoboticsBrno/SmartLeds #20: Compiller error: 'bb' may be used uninitialized in this function [-Werror=maybe-uninitialized]

michaelbrylevskii, 2020-03-03:
Compilation log:
```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
--------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.11.2 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.30300.190916 (3.3.0)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SmartLeds> 1.1.6
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\liba34\SmartLeds_ID1740\Color.o
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp: In constructor 'Rgb::Rgb(Hsv)':
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp:13:39: error: 'bb' may be used uninitialized in this function [-Werror=maybe-uninitialized]
 int down2( int x ) { return x / 255 / 255; }
                                       ^
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp:26:17: note: 'bb' was declared here
     int rr, gg, bb;
                 ^
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp:13:39: error: 'gg' may be used uninitialized in this function [-Werror=maybe-uninitialized]
 int down2( int x ) { return x / 255 / 255; }
                                       ^
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp:26:13: note: 'gg' was declared here
     int rr, gg, bb;
             ^
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp:13:39: error: 'rr' may be used uninitialized in this function [-Werror=maybe-uninitialized]
 int down2( int x ) { return x / 255 / 255; }
                                       ^
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp:26:9: note: 'rr' was declared here
     int rr, gg, bb;
         ^
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp: At global scope:
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp:10:5: warning: 'int {anonymous}::up3(int)' defined but not used [-Wunused-function]
 int up3( int x ) { return x * 255 * 255 * 255; }
     ^
.pio\libdeps\esp32doit-devkit-v1\SmartLeds_ID1740\src\Color.cpp:14:5: warning: 'int {anonymous}::down3(int)' defined but not used [-Wunused-function]
 int down3( int x ) { return x / 255 / 255 / 255; }
     ^
cc1plus.exe: some warnings being treated as errors
*** [.pio\build\esp32doit-devkit-v1\liba34\SmartLeds_ID1740\Color.o] Error 1
```

JarekParal, 2020-03-03:
@michaelbrylevskii  Could you please share your source code which causes this error? Ideally [MCVE](https://stackoverflow.com/help/minimal-reproducible-example).

yaqwsx, 2020-03-03:
@JarekParal if you compile it using Cmake release settings, you get the warnings.

The problem is in the color conversion function; the compiler cannot deduce that hue will fall into one of the 6 options in [the switch](https://github.com/RoboticsBrno/SmartLeds/blob/9d20970c125ea15fcb2021fdd6c750987ed2b51c/src/Color.cpp#L28). The solution is to put there the `default` guard with `__builtin_unreachable`.

michaelbrylevskii, 2020-03-03:
I created PlatformIO project in VS Code.
Build command: C:\Users\<UserName>\.platformio\penv\Scripts\platformio.exe run

main.c
```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
//#include <SmartLeds.h>

#define INT_LED_PIN 2
#define MIC_PIN 32
#define BTN_PIN 13
#define MATRIX_PIN 23
#define MATRIX_LENGTH 256
#define MATRIX_HEIGHT 16
#define MATRIX_WIDTH 16

//SmartLed matrix( LED_WS2812, LED_COUNT, DATA_PIN, CHANNEL, DoubleBuffer );

void app_main()
{
    printf("Hello world!\n");
}
```
platformio.ini
```ini
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf
lib_deps = 
  SmartLeds@1.1.6
monitor_speed = 115200
```

If I changed from `int rr, gg, bb;` to `int rr=0, gg=0, bb=0;` into Color.cpp then compilation successful.

But, if I uncomment two lines in main.c then I've got other compilation error:
`.pio/libdeps/esp32doit-devkit-v1/SmartLeds_ID1740/src/SmartLeds.h:31:18: fatal error: memory: No such file or directory`

yaqwsx, 2020-03-05:
I fixed the issue in f2b0037b7bb3843103ce69870481cc68cba3486c. Please verify it works (the version is now 1.1.7). Note that it might take up to 24 hours to propagate into the PIO repository. Also, I attach a new main function for your example as in the original one there were some issues:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"
#include <SmartLeds.h>

#define INT_LED_PIN 2
#define MIC_PIN 32
#define BTN_PIN 13
#define MATRIX_PIN 23
#define MATRIX_LENGTH 256
#define MATRIX_HEIGHT 16
#define MATRIX_WIDTH 16

SmartLed matrix( LED_WS2812, MATRIX_LENGTH, MATRIX_PIN, 1, DoubleBuffer );

extern "C" void app_main()
{
    printf("Hello world!\n");
}
```

yaqwsx, 2020-03-11:
@michaelbrylevskii have you tested the changes?

yaqwsx, 2020-03-20:
No activity, closing.

closed 2020-03-20 by yaqwsx