RoboticsBrno/SmartLeds #30: Missing include <soc/dport_reg.h> on arduino-esp32-v4.0

chopmann, 2020-12-15:
If building with 
```ini
platform_packages =
    toolchain-xtensa32@3.80200.200512
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0
```

it wont compile because of:
```
SmartLeds.h:111:34: error: 'DPORT_PERIP_CLK_EN_REG' was not declared in this scope
```

it seems that v4.0 is missing libraries. To fix it you need to add:
```c
#include <soc/dport_reg.h>
```
to the Arduino-if.

truebucha, 2021-04-11:
Same for me 
```
#if defined ( ARDUINO )
    extern "C" { // ...someone forgot to put in the includes...
        #include "esp32-hal.h"
        #include "esp_intr_alloc.h"
        #include "driver/gpio.h"
        #include "driver/periph_ctrl.h"
        #include "freertos/semphr.h"
        #include "soc/rmt_struct.h"
        #include "soc/dport_reg.h" << missing
        #include <driver/spi_master.h>
    }
```
won't compile until I added that import.

yaqwsx, 2021-04-14:
@haberturdeur - Could you take care of this? Basically, there should be the include under ifdef.

haberturdeur, 2021-04-14:
@truebucha on what version of esp32 platform are you trying to run?

truebucha, 2021-04-17:
Expressif32 3.2.0 through platformio

haberturdeur, 2021-04-17:
@truebucha Can you try adding this instead, I need to test it, but I can't get the new arduino running... 
```
        #include "esp_idf_version.h"
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 0, 0)
        #include "soc/dport_reg.h"
#endif
```

commit 6f54b2fa (2021-04-27): Fix missing include

closes #30