RoboticsBrno/SmartLeds #35: Assert statement causes compilation error with default ESP32 build in platformIO

GrantRolls, 2021-06-24:
Assert statement causes compilation error with default ESP32 build in platformIO

```
In file included from C:/Users/Grant/.platformio/packages/framework-espidf/components/newlib/platform_include/assert.h:22,
                 from c:\users\grant\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\8.4.0\cassert:44,
                 from lib/SmartLeds/src/Color.cpp:4:
lib/SmartLeds/src/Color.cpp: In function 'int iRgbSqrt(int)':
lib/SmartLeds/src/Color.cpp:15:57: error: left operand of comma operator has no effect 
[-Werror=unused-value]
     assert(("sqrt input should be non-negative", num >= 0));
                                                         ^
lib/SmartLeds/src/Color.cpp:16:59: error: left operand of comma operator has no effect 
[-Werror=unused-value]
     assert(("sqrt input should no exceed 16 bits", num <= 0xFFFF));
                                                           ^~~~~~
cc1plus.exe: some warnings being treated as errors
*** [.pio\build\esp32-evb\lib287\SmartLeds\Color.o] Error 1
```

haberturdeur, 2021-06-24:
Should be ok now, can you confirm please?

closed 2021-06-24 by haberturdeur

yaqwsx, 2021-06-27:
I would also suggest removing the extra pair of braces (they are only confusing).

commit 0003d4c1 (2021-06-24): Fix assert warning

Fixes #35